DISTORTEC/distortos #2: custom toolchain

ilg-ul, 2015-01-05:
we did some further tests, and (with the linux toolchain downloaded from https://sourceforge.net/projects/bleeding-edge/files/141220/), when using -Og, the function 
```void ThreadControlBlock::threadRunner(ThreadControlBlock& threadControlBlock)``` 
is excessively optimised to a while(1) loop.

for the same code, using the latest launchpad compiler, -Og seems ok, the generated code for the above function is correct.

I have no idea what differs between the two toolchains, but apparently they are build differently.

regards,

Liviu

p.s. as a suggestion for your builds: use -Og -fno-move-loop-invariants for the debug configuration and -Os for the release configuration.

KamilSzczygiel, 2015-01-05:
> we did some further tests, and (with the linux toolchain downloaded from https://sourceforge.net/projects/bleeding-edge/files/141220/), when using -Og, the function 
void ThreadControlBlock::threadRunner(ThreadControlBlock& threadControlBlock) 
is excessively optimised to a while(1) loop.

Hmm... In here it looks perfectly fine. Maybe you just misinterpreted the data?
```
08000e80 <distortos::scheduler::ThreadControlBlock::threadRunner(distortos::scheduler::ThreadControlBlock&)>:
/*---------------------------------------------------------------------------------------------------------------------+
| private static functions
+---------------------------------------------------------------------------------------------------------------------*/

void ThreadControlBlock::threadRunner(ThreadControlBlock& threadControlBlock)
{
 8000e80:	b508      	push	{r3, lr}
	threadControlBlock.run();
 8000e82:	6803      	ldr	r3, [r0, #0]
 8000e84:	681b      	ldr	r3, [r3, #0]
 8000e86:	4798      	blx	r3
	getScheduler().remove();
 8000e88:	f000 f95e 	bl	8001148 <distortos::scheduler::getScheduler()>
 8000e8c:	f7ff fdea 	bl	8000a64 <distortos::scheduler::Scheduler::remove()>

/*---------------------------------------------------------------------------------------------------------------------+
| private static functions
+---------------------------------------------------------------------------------------------------------------------*/

void ThreadControlBlock::threadRunner(ThreadControlBlock& threadControlBlock)
 8000e90:	e7fe      	b.n	8000e90 <distortos::scheduler::ThreadControlBlock::threadRunner(distortos::scheduler::ThreadControlBlock&)+0x10>
 8000e92:	bf00      	nop
```

The real problem is indeed caused by the optimization level, but this function has nothing to do with it. It's the same problem that you reported in comment to issue #1 - with `-Og` the stack size for one of the tests is too small, this leads to stack overflow, which in turn leads to a failure in distortos operation. The only difference is that this time it's not a crash, but failure to terminate test thread, which results in reaching endless `while (1);` loop in `distortos::scheduler::ThreadControlBlock::threadRunner()`, which normally is never reached.

If you want to use this optimization level, then you need to set stack size for all 3 subtests of `MutexPriorityInheritanceOperationsTestCase` to 512 bytes (there are 3 constants named `testThreadStackSize` in the `MutexPriorityInheritanceOperationsTestCase.cpp` file). Alternatively 384 bytes is enough for the first subtest - `testBasicPriorityInheritance()`, but two next subtests require 512 bytes.

With these change (and the change to optimization level) whole OS test finishes correctly, but I'm pretty certain that there are some more stack overflows, they just don't cause any (visible) problems... The OS works for all optimization levels, but the sizes of stacks in OS test application are good only for `-O2`. The stack sizes may be big enough for other optimization levels (like `-Os` or `-O3`), but there is no guarantee - I check them only for `-O2`, as that's the optimization level I'm using.

> p.s. as a suggestion for your builds: use -Og -fno-move-loop-invariants for the debug configuration and -Os for the release configuration.

This all comes down to personal preferences, but for me `-O2` is the "perfect" level. It is a right balance between speed and size. What is most problematic in embedded is the RAM, and using `-Os` over any other level gives nothing in this regard. For example with the problematic test we're discussing here - `MutexPriorityInheritanceOperationsTestCase` and `testPriorityChange()` subtest - `-O2` uses 140 bytes of stack, while `-Os` - 144-152, so `-O2` wins (; (and `-Og` uses a little over 400)

On the other hand using `-Os` over `-O2` makes the binary about ~9kB smaller, but we have to remember that distortos doesn't aim to work on low-end small microcontrollers with little memory available. For me these 9kB are not worth the performance that would be lost, and looking at the assembly generated for `-Os` it is really _NOT_ optimal...

Of course anyone can use whichever level he/she prefers - distortos should work fine on all of them. On my "todo" list I have a task to change stack sizes of all test threads so that the OS test would run on any optimization. I had no motive for doing that (I use only `-O2`), but now I may do that sooner (maybe even today).

ilg-ul, 2015-01-05:
> Maybe you just misinterpreted the data?

yes, it might be so, I checked only at 8000e90 and it looked like the entire function is there, did not check above.

anyway, when using the launchpad toolchain the test completed, when using your toolchain it ended in that loop. I understand now this is not relevant, due to the stack limitation.

KamilSzczygiel, 2015-01-05:
> anyway, when using the launchpad toolchain the test completed, when using your toolchain it ended in that loop. I understand now this is not relevant, due to the stack limitation.

Probably a pure coincidence, but I'll run it with launchpad toolchain too to see how it behaves.

KamilSzczygiel, 2015-01-05:
As mentioned in the commit above, the "OS test" works correctly with following optimization levels: `-O1`, `-O2`, `-O3`, `-Os`, `-Og`. Using `-O0` at this moment is not possible, because stack overflows are only one part of the problem - the smaller one. With `-O0` the code runs so slow (nothing is inlined, so template instantiations sometimes go really deep) that some assumptions of many tests are violated. For example in `ThreadSleepForTestCase` it takes about 7 ticks for all 10 test threads to actually go to sleep, while it is assumed that all of that code would fit into less than 1 tick - this is needed for proper calculation of sleep duration precision. I didn't find any easy way to fix that other than modifying default configuration to decrease tick rate, but I would prefer not to do that, because I think there will be more problems related to execution speed in other tests...

In future proper support for boards will be implemented along with drivers for the chip peripherals. This will allow to set the clock of the device to higher rate, which would probably make the test usable with `-O0`, so I'll increase the sizes of stack then.

I tested with my own bleeding-edge-toolchain and with "official" toolchain from launchpad - both work correctly.

What is intersting is that distortos (at this stage - see below) does not pull exception handling with "official" toolchain, so at this moment it doesn't make much difference which toolchain is used.

This is however only temporary - as soon as more advanced features will be implemented (anything that requires dynamic allocation) exception handling will be pulled again, increasing the size of executable by ~70kB (and static RAM usage by ~3kB).

Anyway - if you still experience this problem, please reopen the issue (or open a new one if this will be something slightly different). I think the problem mentioned in this issue is solved, that's why I consider this one fixed.