DISTORTEC/distortos #3: question:

geraldstanje, 2015-03-31:
hi, is there a plan to support arm cortex a9?

Thanks,
Gerald

KamilSzczygiel, 2015-03-31:
There's no technical reason against such support - ARM Cortex-A9 has all the required features, enough memory and processing power, the toolchain for such chip supports all the important C++11 features. However I'm not sure such chip would be a good target for this project. You could probably port distortos to Intel Core i7 and run it on your PC, but this just makes little sense (; This project is intended mainly for higher-end microcontrollers with no support for memory management unit, while ARM Cortex-A9 is an application processor with MMU, capable of running Linux or other general purpose operating system. It seems like an "overkill"...

However the line between microcontroller and microprocessor is getting more and more blurred - for example ARM Cortex-A5 or ARM Cortex-M7 are examples that can be considered something in between a microcontroller and microprocessor.

I'm not saying "no" - I just have no experience with ARM Cortex-A and I'm not sure such support in distortos would make sense - either the project wouldn't use all features of such powerful core (like MMU) or it would make the whole source code much more complex to include features used only by such target (which would be impossible to enable on - for example - ARM Cortex-M). Both of these options can be seen as a problem by people interested in using it - people using ARM Cortex-A would be disappointed by lack of MMU support (in first option) or people using ARM Cortex-M would be disappointed by complexity of the code base...

What do you think about that?

geraldstanje, 2015-03-31:
which compiler and dev. environment do you currently use?

you can configure the arm cortex a9 for bare metal!

KamilSzczygiel, 2015-03-31:
> which compiler and dev. environment do you currently use?

I'm using my own compilation of GCC - it's called "bleeding-edge-toolchain" ( https://sourceforge.net/projects/bleeding-edge/ , there's also a link to an article on my website about that toolchain variant in the README.md file in the repository). You can also use original GCC from ARM ( https://launchpad.net/gcc-arm-embedded ), but bleeding-edge-toolchain is much better choice (it has disabled C++ exceptions and some options of newlib are better suited for RTOS) [I can elaborate more about that if you want (; ]. Both of these toolchains support ARM Cortex-A targets. I'm using that in Eclipse on Linux, but it works the same way on Windows. To debug that I'm using OpenOCD.

> you can configure the arm cortex a9 for bare metal without any linux! im not planning to use the MMU on bare metal...

I know, it's just that if you use ARM Cortex-A without an MMU, you basically get a fast and expensive (and probably less responsive) version of ARM Cortex-M... ARM Cortex-M can also run from external memory or have SDRAMs, some chips have FPU (single precision). OK, another thing that ARM Cortex-A has and ARM Cortex-M doesn't is the NEON engine...

geraldstanje, 2015-04-24:
how can you have exceptions in a gcc toolchain? is there a c++ compiler for the arm cortex a9?

KamilSzczygiel, 2015-04-24:
You can use the one provided by ARM - https://launchpad.net/gcc-arm-embedded - it has exceptions enabled and it has libraries for ARM Cortex-A chips. Its newlib configuration is not perfect for an RTOS, but mainly because of RAM usage, and I guess RAM is not an issue for ARM Cortex-A9 (; From what I see, there are no libs for ARM mode - only for Thumb mode...

There are also linaro toolchains - https://wiki.linaro.org/Cycles/1503/Release#Toolchain_Components - they provide some bare-metal toolchain components, so I guess it's also an option, but probably more complex then the toolchain from launchpad.

Take note that distortos is probably not "ready" for exceptions...

If you browsed through distortos' sources, would you mind sharing some opinions with me? I'd very much like to hear what you think about the overall idea, whether there are things you think should be done differently (API, algorithms, implementations, ...) - anything that comes to your mind (; I'm also very interested in the progress of your port to ARM Cortex-A - if you started that already. If you don't want to spam this "issue" page, you can reach me via e-mail - on the websites listed in my profiles - www.freddiechopin.info or www.distortec.com - there are contact forms.

closed 2015-09-09 by KamilSzczygiel