DISTORTEC/distortos #18: Compilation error for arm-none-eabi-gcc 4.9.3

gkostka, 2016-01-11:
Hi,

I discovered some compilation errors for 4.9.3 compiler (taken from Debian sid packages). Here are some trivial fixes:
https://github.com/DISTORTEC/distortos/pull/17

Here is a hard one for pure C programmer;)
```
In file included from include/distortos/internal/synchronization/MessageQueueBase.hpp:20:0,
                 from source/synchronization/MessageQueueBase.cpp:12:
include/estd/SortedIntrusiveForwardList.hpp: In instantiation of 'constexpr estd::SortedIntrusiveForwardList<Compare, T, NodePointer, U>::Implementation::Implementation(const Compare&) [with Compare = distortos::internal::MessageQueueBase::DescendingPriority; T = distortos::internal::MessageQueueBase::Entry; estd::IntrusiveForwardListNode T::* NodePointer = &distortos::internal::MessageQueueBase::Entry::node; U = distortos::internal::MessageQueueBase::Entry]':
include/estd/SortedIntrusiveForwardList.hpp:75:27:   required from 'constexpr estd::SortedIntrusiveForwardList<Compare, T, NodePointer, U>::SortedIntrusiveForwardList(const Compare&) [with Compare = distortos::internal::MessageQueueBase::DescendingPriority; T = distortos::internal::MessageQueueBase::Entry; estd::IntrusiveForwardListNode T::* NodePointer = &distortos::internal::MessageQueueBase::Entry::node; U = distortos::internal::MessageQueueBase::Entry]'
source/synchronization/MessageQueueBase.cpp:142:18:   required from here
include/estd/SortedIntrusiveForwardList.hpp:278:26: error: too many initializers for 'distortos::internal::MessageQueueBase::DescendingPriority'
     intrusiveForwardList{}
                          ^
```

FreddieChopin, 2016-01-11:
> Here is a hard one for pure C programmer;)

Believe it or not, but I don't understand why this error is there... This should not happen, yet it does. Maybe it's some kind of compiler bug, no idea... The fix is trivial - `distortos::internal::MessageQueueBase::DescendingPriority` struct needs a constexpr default constructor. I have that fixed locally and I'll push that in a moment.

I'll also push more fixes for missing cerrno

> Maybe it would be good idea to build distortos on travis using more traditional toolchains (Linaro or Launchpad). Not everyone likes bleeding edge approach :)

The problem is that these toolchains don't have optimal configuration for distortos. There are two major things that are different between them and bleeding-edge-toolchain:

1. C++ exceptions - bleeding-edge-toolchain has them disabled during compilation of libstdc++, with the toolchains you mention you can expect the size of executable to increase A LOT (like a few dozen kB); in some cases this can be avoided, but current test application will pull C++ exceptions for sure
2. bleeding-edge-toolchain uses "small" _reent structure, while typical newlib configuration uses "large" version; This doesn't make much difference for single-threaded application and it doesn't make a difference for a RTOS which ignores _reent in newlib (*); in distortos every thread has its own _reent, so it really makes a difference whether this struct is ~100b (in "small" version) or ~1kB (in "large" version)...

Executable compiled with these toolchains would work correctly (until a C++ exception is thrown, which would probably result in a crash), but it would be really big and use a lot of RAM...

I plan to add one (4.9) or two (4.8 and 4.9) previous versions of GCC to travis check.

\* - if you use single _reent instance in multi-threaded application you cannot expect many of standard library functions to work correctly, which includes things like rand(), asctime(), strtok(). Additionally using errno becomes impossible, as this variable is part of _reent structure.

FreddieChopin, 2016-01-11:
OK, all the compilation problems with GCC 4.9 should be fixed now. I've also added this version of GCC to travis.

If you encounter any more problems then either reopen this issue or create new one.

closed 2016-01-11 by FreddieChopin