DISTORTEC/distortos #21: Plans to support the STM32F7 series?

rahmanih, 2016-06-09:
Hi,
Just to know whether there are plans or not to support [the F7 series](http://www.st.com/content/st_com/en/products/microcontrollers/stm32-32-bit-arm-cortex-mcus/stm32f7-series.html?querycriteria=productId=SS1858)?

regards
Haithem.

FreddieChopin, 2016-06-09:
Hello Haithem!

Yes! Support for STM32F7 is planned, as these chips are a natural target for distortos. I already have the [32F746GDISCOVERY](http://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-eval-tools/stm32-mcu-eval-tools/stm32-mcu-discovery-kits/32f746gdiscovery.html) board (I could also get NUCLEO-F746ZG  or NUCLEO-F767ZI if that would be preferred), so I actually can start adding such support right away.

Would you like to start some project with STM32F7? In that case I could start adding it - I think that in a couple of days it would be ready.

rahmanih, 2016-06-09:
Great news!
I'm not in a harry, please take your time ;).

thanks and regards