DISTORTEC/distortos #22: Use readlink instead of realpath in generateKconfig.sh?

cw2, 2016-10-10:
Hello,

I have managed to successfully build distortos in [Bash on Ubuntu on Windows 10](https://msdn.microsoft.com/en-us/commandline/wsl/about) and I would like to ask if you could consider using `readlink` instead of `realpath --relative-base=.` in `scripts\generateKconfig.sh` ?

There is a slight problem with _coreutils_ package that comes with Ubuntu on Windows release, namely its `realpath` command does not support `--relative-base` option, which causes the build to fail. I am aware of the official instructions to use GNU Coreutils and the existing confusion with the other _coreutils_ [1], but it seems `realpath` is the only command needed to build distortos (I have commented out the line in the .sh script and there have been no other errors) and replacing it would remove the GNU Coreutils dependency, thus making the whole process easier (?)

[1] [What's the difference between “realpath” and “readlink -f”](http://unix.stackexchange.com/a/136527)

Thanks,
---
CW2

FreddieChopin, 2016-10-10:
Are you using Windows 10 normally? Have you considered using MSYS2, adding its folders to system's PATH and building the project normally, via Windows command prompt (or in any IDE of your choice)? I'm asking because maybe it would be easier to just use something else, and the approach with MSYS2 works fine (at least it worked a while ago in Windows 7, with which I test the build).

Generally `realpath` is used only to "normalize" the path, so the script would work even if you configure `output` to be sth as bizarre as `././././././//////////output//////////////////////////` or whatever you like. Root cause for such "normalization" is that that both `find` invocations below are designed to print the files with matching names - for example "Kconfig-chipOptions" - but ignore the file that will be generated by the script, by just ignoring everything in `$output` folder. That's the `... -path "./$output" -prune ...` part. The main problem with this line is that it really doesn't like anything other than a pretty name of folder with no extra decorations other than "./" at the front. For example:

```
$ find -path "./output" -prune -o -name Kconfig-chipOptions -printf 'source "%p"\n' | sortsource "./source/chip/STM32/STM32F0/Kconfig-chipOptions"
source "./source/chip/STM32/STM32F1/Kconfig-chipOptions"
source "./source/chip/STM32/STM32F4/Kconfig-chipOptions"
$ find -path "./././././././////output" -prune -o -name Kconfig-chipOptions -printf 'source "%p"\n' | sort
source "./output/Kconfig-chipOptions"
source "./source/chip/STM32/STM32F0/Kconfig-chipOptions"
source "./source/chip/STM32/STM32F1/Kconfig-chipOptions"
source "./source/chip/STM32/STM32F4/Kconfig-chipOptions"
```

The main difference that I see here is that `realpath` gives you relative path with absolutely no decorations, while readlink would give you an absolute path. So in the later case I think that either the `find` invocations would have to change (maybe that's possible, maybe not) or you'd also have to use `pwd` and `expr` (or `sed` or whatever) to remove the working directory from the returned path. Another options is to just perform some string analysis and modifications, but when I wrote that script and browsed the net for solutions, all the commands that I saw were extremely horrible collections of dozens of slashes, so I just chose `realpath` (;

Generally the idea is to meet both of these requirements:
- the script should work with any reasonable path provided - with "./" at the beginning or without, with trailing slash or without,
- both `find` invocations have to ignore all files that are located in the provided path (`$output`).

As for the dependency on Coreutils, I think that `realpath` is not the only required tool that is part of this package - for example there's also `cat`, `sort`, `rm`, `mkdir`, `echo` and so on...

FreddieChopin, 2016-10-10:
Please check the current version of the script and let me know whether it works for you now. I've removed call to `realpath` - which is not a standard solution - but I did not use `readlink -f` too - it seems that the `-f` switch is not supported on all platforms... So I've used a combination of `cd` and `pwd`, which basically gives the same effect as `readlink -f`. The only difference is that this solution does not resolve symbolic links, but this is not an issue here.

cw2, 2016-10-11:
The current version of the script works fine.

Thanks.

commit 08a4d476 (2016-10-10): Don't use "realpath" in generateKconfig.sh

"realpath" is not a standard solution - the "--relative-base=..." switch
is not supported on all platforms. Normalize path with "cd" and "pwd",
which seems slightly more complicated, but is more portable.

There is no point in having the argument optional, so now the script
exits with an error if no argument is given.

Fixes #22