DISTORTEC/distortos #23: Diagnostic output?

cw2, 2016-10-10:
I hope I did not miss anything, but it seems there is no support for diagnostic output in the current implementation of distortos? Are there any plans to add for example Instrumentation Trace Macrocell (ITM) output (I dare to say in printf()-like redirection)?

_(I would consider a contribution, but at the moment I am wrapping my head around embedded C++11 code...)_

FreddieChopin, 2016-10-10:
You did not miss anything (; There a lot of nice features missing and - as you imagine - my plans for future are really long. Diagnostic outputs and support for `printf()` are both in these plans. The only problem here is time... Recently the project slowed down a little bit, because we had a "small" renovation in the house for almost whole summer.

Actually I haven't used ITM myself, my plans for diagnostic output currently are pretty blurry - I have some ideas, but I did not start anything yet. So if you'd like to contribute something, even if that would be an idea posted here or to the mailing list - that would be great!

Don't be afraid of C++11 and embedded - such combination causes no trouble as long as you use  toolchain with disabled C++ exceptions (*). But I guess this all comes down to preferences - some people prefer C++, some people will never use anything else than pure C.

(*) - I also have a plan to remove the need for special toolchain with disabled C++ exceptions and specific newlib configuration, but this also comes down to the problem of time...

cw2, 2016-10-10:
I understand completely, I have a ToDo list(s) too and only limited time for embedded endeavors (hopefully, The Winter is coming : )

Regarding the ITM, it is rather simple - as you are already using CMSIS headers, it basically means just enabling the peripheral and stimulus port (usually 0) and call `ITM_SendChar(...)` defined in core_cmX.h.

The main benefit of ITM is that for STMicro development boards that have ST-LINK debugger it does not require any additional hardware (no USB<->Serial dongle or such), because its output is routed through ST-LINK and can be viewed directly in STM32 ST-Link utility or OpenOCD (and some IDEs).

I am not afraid of C++11 and embedded, the use of C++ is one of the main reasons for my interest in this project. I need some time to get used to those new C++11 constructs and GCC-specific extensions (the code just looks weird, in comparison to what would be considered a 'usual embedded code' ;)

I am also aware of your Bleeding Edge Toolchain and I have some experience with building custom GCC (I use GNU ARM Embedded Toolchain with newlib-nano that has enabled long-long and long-double support, for a [project you might never heard of](https://github.com/cw2/netmf-interpreter), where I have participated in firmware development efforts, recently for STM32F7 (work in progress)).