DISTORTEC/distortos #24: Tickless mode?

cw2, 2016-10-17:
I am just wondering if support for tickless mode is on your TODO list?

FreddieChopin, 2016-10-17:
It is (; I had this feature on my mind for a long time, but... yes... time (; As you probably imagine, the list is pretty long and sometimes it's hard for me to decide what's more important and what can wait. So far I've just followed the input I got (here, in private, on forums) and the requirements for the projects I'm working on. Generally there are a lot of "fields" on which *distortos* could be improved - for example it could support more chips (like STM32F7, STM32F2, ...), more chip families (for example NXP, Kinetis, TI, ...), more architectures, it could have more general features (like tickless mode), more drivers for peripherals (I2C, CAN, ETH, ...), more generic drivers (memory card, external chips, ...), integration with popular libraries (lwIP, FatFS, ...) and so on, and so on...

Ok, but back to the point... If there's any feature that's very important for you, something that blocks a nice project which could use *distortos*, or just something you consider a "must-have" for RTOS - just let me know, whatever it is - I'll bump its priority on my list much higher. If you have a timeframe - let me know too, I'll see what I can do.

cw2, 2016-10-17:
Thanks for such fast response.

My project is at very early stage, I am basically gathering information and components that could be used. In this particular case, the lack of tickless mode is not really a blocker, technically the system will work without it, but it will have serious drawback, i.e. no low power operation (deep sleep).

I guess adding such feature in still relatively early phase of development (?) bears less risk, in terms of core functionality that would have to be changed. Unfortunately, my knowledge of distortos is still rather limited. I can imagine adding more peripheral or generic drivers, or even chips or new families (I have already done similar work), but I have no idea about changing core mechanisms (i.e. whether adding tickless mode means rewriting majority of the existing code or simply overriding one member function ;)

Nonetheless, tickless mode is very important in certain scenarios (connected sensors and such), and modern RTOS-es have it (not to mention it's a nice marketing item).