DISTORTEC/distortos #34: Combined interrupts for USART4 and USART5

CezaryGapinski, 2017-05-05:
Some of the microcontrollers (like in STM32L0's familly) have common interrupt for USART4 and USART5.
Patch to fix thiss issue is here:
https://github.com/CezaryGapinski/distortos/commit/484cca388ef287ba487c4fe44bbe78a6cefc6648

Change is on my develop branch for STM32L0, but I think that will be easy to get "cherry-pick" from this commit to the new branch direct from "master" branch, but I would like to ask you what do you thinking about that changes?

FreddieChopin, 2017-05-05:
I did not check your patch in detail, but I assume you've seen this - https://github.com/DISTORTEC/distortos/blob/master/source/chip/STM32/peripherals/USARTv2/STM32-USARTv2-uarts.cpp ? Did you base your change on code from STM32's USARTv2?

FreddieChopin, 2017-05-05:
Please ignore what I wrote before (;

FreddieChopin, 2017-05-05:
I think the change is OK. We don't have to cherry-pick it to the master now - it can stay in your STM32L0 branch and get merged together with the rest of the changes. Unless you want to make your branch shorter? Just let me know if you do and I'll cherry-pick this patch in the evening.

CezaryGapinski, 2017-05-05:
There is no problem to leave it on my branch. I would like just change the sequence of last two commits and put "Add support for USARTv2 to STM32L0's Kconfig menus" on "Add combined interrupts for USART4-5 in USARTv2". That is more logic sequence in my opinion, because then
`select CHIP_STM32_USARTV2_USART4_USART5_COMBINED_INTERRUPT if CHIP_STM32L07 || CHIP_STM32L08 `
condition in Kconfig-stm32ChipFamilyChoices won't cause confusion and is the result of previous changes.

I tested the USART driver with your test program and looks like it also works fine ;).

FreddieChopin, 2017-05-05:
Sure, change the order to make the sequence of changes more logical - it would be indeed wrong (or at least very confusing) to implement in first commit something which depends on the code introduced in second commit.

closed 2017-05-21 by FreddieChopin