DISTORTEC/distortos #59: Problem following "Getting Started" in Windows

syntroniks, 2021-02-27:
```
Windows 10 Pro Build 21286
~\Documents\distortos\build [master ≡ +1 ~0 -0 !]> git describe
v0.7.0-299-g53ee25952

~\Documents\distortos\build [master ≡ +1 ~0 -0 !]> cmake --version
cmake version 3.19.5

CMake suite maintained and supported by Kitware (kitware.com/cmake).

~\Documents\distortos\build [master ≡ +1 ~0 -0 !]> ninja --version
1.10.1

~\Documents\distortos\build [master ≡ +1 ~0 -0 !]> arm-none-eabi-gcc --version
arm-none-eabi-gcc.exe (GNU Arm Embedded Toolchain 10-2020-q2-preview) 10.1.1 20200529 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
=>
```
~\Documents\distortos\build [master ≡ +1 ~0 -0 !]> cmake .. -DCMAKE_TOOLCHAIN_FILE=..\source\board\ST_STM32F4DISCOVERY\Toolchain-ST_STM32F4DISCOVERY.cmake -GNinja
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:17 (project):
  No CMAKE_C_COMPILER could be found.
...
```

Previously it was selecting clang (until I uninstalled it).

It _looks_ as if the toolchain file isn't being ran, but when I mis-spell it, CMake complains about 'can't find file'. That said, adding `message(...)` calls in the toolchain file doesn't result in output.

This works fine in wsl.

syntroniks, 2021-02-27:
As a workaround, I've been experimenting in Visual Studio CMake projects.

Choosing an Existing Cache for `CMakeSettings.json` that pulls from a .bat file (that uses `CMakePresets.json`) things are workable until MS upgrade VS and VScode.

Looking forward to getting started with distortos.

FreddieChopin, 2021-02-28:
> Previously it was selecting clang (until I uninstalled it).

Have you deleted the `build` folder after all previous failed attempts and tried with an empty one? This is very important! CMake is pretty stupid in this regard - if there is ANY error during the initial "folder configuration" run, then the whole directory is completely unrecoverable, it MUST be deleted and you have to start with an empty one.

One _VERY_ important thing. *distortos* is meant to be built with a "bleeding-edge-toolchain", not the toolchain provided by ARM. Even if it will build, there are ZERO guarantees that it will work correctly when using that "official" toolchain from ARM. Even if it will work, it WILL use much more RAM in all possible cases!

https://github.com/FreddieChopin/bleeding-edge-toolchain

You can download Windows builds of this toolchain from here:
https://freddiechopin.info/en/download/category/11-bleeding-edge-toolchain

If the problem still persists, please attach all the `*log` files that you can find in the build folder (especially `build/CMakeFiles/CMakeError.log` and `build/CMakeFiles/CMakeOutput.log`).

As a side note, please also try your command with forward slashes (in `-DCMAKE_TOOLCHAIN_FILE=`, maybe there's some typical Linux/Windows-slash-related issue hidden somewhere...